kaosat-dev/adafruit-i2c-pwm-driver #3: Problem to use "adafruit-i2c-pwm-driver" in node.js

tilljohan, 2016-05-06:
I have problem to install/use "adafruit-i2c-pwm-driver" in node.
I have have tried 4.0.0, 5, 6.1.0
It worked great in node 0.10.3

Maybe this?
Latest version for i2c@0.2.1

npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "adafruit-i2c-pwm-driver"
npm ERR! node v6.1.0
npm ERR! npm  v3.8.9
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.1.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.1.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm in

kaosat-dev, 2016-10-08:
Hi @tilljohan ! Very sorry for the extremely late reply !
I will be solving this (and other issues) and updating the library in the next few weeks.

EMCP, 2016-10-31:
not sure if related, but on `npm i` I'm getting the following:

```
i2c.target.mk:90: recipe for target 'Release/obj.target/i2c/src/i2c.o' failed
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
make: Leaving directory '/home/pi/Dev/kaosat-dev/adafruit-i2c-pwm-driver/node_modules/i2c/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.26-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/Dev/kaosat-dev/adafruit-i2c-pwm-driver/node_modules/i2c
gyp ERR! node -v v7.0.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
adafruit-i2c-pwm-driver@0.0.2 /home/pi/Dev/kaosat-dev/adafruit-i2c-pwm-driver
├── coffee-script@1.3.3 
├─┬ sleep@1.1.8 
│ └── mkdirp@0.3.5 
└── underscore@1.2.4 

npm ERR! Linux 4.4.26-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "i"
npm ERR! node v7.0.0
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.1.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.1.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs i2c
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/Dev/kaosat-dev/adafruit-i2c-pwm-driver/npm-debug.log

```

kaosat-dev, 2016-11-06:
@tilljohan this should now be fixed !
but beware, there were some api changes in the process (please see the updated readme for more details).

tilljohan, 2016-11-07:
Yes, its working with later builds.

kaosat-dev, 2016-11-07:
@tilljohan  thanks for confirming, closing this issue

closed 2016-11-07 by kaosat-dev