kaosat-dev/adafruit-i2c-pwm-driver #5: project doesn't install on rasbpian

EMCP, 2016-11-06:
Raspberry Pi 2 running Raspbian Lite

Output 

```
pi@raspberrypi:~/Dev/kaosat-dev/adafruit-i2c-pwm-driver $ npm i

> i2c@0.1.1 install /home/pi/Dev/kaosat-dev/adafruit-i2c-pwm-driver/node_modules/i2c
> node-gyp rebuild

make: Entering directory '/home/pi/Dev/kaosat-dev/adafruit-i2c-pwm-driver/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
../src/i2c.cc:17:32: error: ‘Arguments’ does not name a type
 Handle<Value> SetAddress(const Arguments& args) {
                                ^
In file included from /home/pi/.node-gyp/6.9.1/include/node/node.h:42:0,
                 from ../src/i2c.cc:1:
/home/pi/.node-gyp/6.9.1/include/node/v8.h: In function ‘v8::Handle<v8::Value> SetAddress(const int&)’:
/home/pi/.node-gyp/6.9.1/include/node/v8.h:915:13: error: ‘v8::HandleScope::HandleScope()’ is protected
   V8_INLINE HandleScope() {}
             ^
../src/i2c.cc:18:15: error: within this context
   HandleScope scope;
               ^
../src/i2c.cc:20:16: error: invalid types ‘const int[int]’ for array subscript
   addr = args[0]->Int32Value();
                ^
../src/i2c.cc:24:16: error: ‘class v8::HandleScope’ has no member named ‘Close’
   return scope.Close(Undefined());
                ^
../src/i2c.cc:24:32: error: too few arguments to function ‘v8::Local<v8::Primitive> v8::Undefined(v8::Isolate*)’
   return scope.Close(Undefined());
                                ^
In file included from /home/pi/.node-gyp/6.9.1/include/node/node.h:42:0,
                 from ../src/i2c.cc:1:
/home/pi/.node-gyp/6.9.1/include/node/v8.h:324:27: note: declared here
   friend Local<Primitive> Undefined(Isolate* isolate);
                           ^
../src/i2c.cc: At global scope:
../src/i2c.cc:27:26: error: ‘Arguments’ does not name a type
 Handle<Value> Scan(const Arguments& args) {
                          ^
In file included from /home/pi/.node-gyp/6.9.1/include/node/node.h:42:0,
                 from ../src/i2c.cc:1:
/home/pi/.node-gyp/6.9.1/include/node/v8.h: In function ‘v8::Handle<v8::Value> Scan(const int&)’:
/home/pi/.node-gyp/6.9.1/include/node/v8.h:915:13: error: ‘v8::HandleScope::HandleScope()’ is protected
   V8_INLINE HandleScope() {}
             ^
../src/i2c.cc:28:15: error: within this context
   HandleScope scope;
               ^
../src/i2c.cc:31:58: error: invalid types ‘const int[int]’ for array subscript
   Local<Function> callback = Local<Function>::Cast(args[0]);
                                                          ^
../src/i2c.cc:32:38: error: invalid conversion from ‘int’ to ‘v8::Isolate*’ [-fpermissive]
   Local<Array> results(Array::New(128));
                                      ^
In file included from /home/pi/.node-gyp/6.9.1/include/node/node.h:42:0,
                 from ../src/i2c.cc:1:
/home/pi/.node-gyp/6.9.1/include/node/v8.h:3039:23: note: initializing argument 1 of ‘static v8::Local<v8::Array> v8::Array::New(v8::Isolate*, int)’
   static Local<Array> New(Isolate* isolate, int length = 0);
                       ^
../src/i2c.cc:33:45: error: too few arguments to function ‘v8::Local<v8::Primitive> v8::Null(v8::Isolate*)’
   Local<Value> err = Local<Value>::New(Null());
                                             ^
In file included from /home/pi/.node-gyp/6.9.1/include/node/node.h:42:0,
                 from ../src/i2c.cc:1:
/home/pi/.node-gyp/6.9.1/include/node/v8.h:325:27: note: declared here
   friend Local<Primitive> Null(Isolate* isolate);
                           ^
[... 582 lines cut ...]
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.1.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.1.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs i2c
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/Dev/kaosat-dev/adafruit-i2c-pwm-driver/npm-debug.log
pi@raspberrypi:~/Dev/kaosat-dev/adafruit-i2c-pwm-driver $ node --version
v6.9.1
pi@raspberrypi:~/Dev/kaosat-dev/adafruit-i2c-pwm-driver $ npm --version
3.10.8

```

EMCP, 2016-11-06:
turns out, I may need to follow this : 

https://github.com/kelly/node-i2c

EMCP, 2016-11-06:
upgrading to the latest `i2c@0.2.3` made the errors go way.. please update the package

EMCP, 2016-11-06:
here's my `package.json` that built fully

```
{
  "name": "adafruit-i2c-pwm-driver",
  "version": "0.0.2",
  "description": "node.js /i2c control for the Adafruit PWM servo driver",
  "main": "main.js",
  "author": "Mark Moissette",
  "license": "MIT",
  "keywords": [
    "raspberry",
    "pi",
    "adafruit",
    "i2c",
    "pwm",
    "servo"
  ],
  "repository": {
    "type": "git",
    "url": "https://github.com/kaosat-dev/adafruit-i2c-pwm-driver"
  },
  "bugs": {
    "url": "https://github.com/kaosat-dev/adafruit-i2c-pwm-driver/issues"
  },
  "dependencies": {
    "underscore": "1.2.4",
    "coffee-script": "1.3.3",
    "i2c": "0.2.3",
    "sleep": "~3.0.0"
  }
}
```

kaosat-dev, 2016-11-06:
@EMCP I just updated the npm package , in addition to the changes here on github: given that your  changes are very close to mine I hope you will not have problem with the new version :)
Thanks for the help !

closed 2016-11-06 by EMCP